rotorflight/rotorflight-configurator #23: Refactor Servos tab

rotorflight, 2022-12-28:
The servo configuration has been totally refactored in the firmware. The parameters are reflecting the internal function of the firmware, and should not be exposed to the end-user directly. Instead, the Servo tab should offer more user-oriented view for the configuration.

The best idea so far is to have the following parameters offered for each servo:

- Servo type: 1520us or 760us or _custom_
  - _Custom_ can't be selected by the user. It is selected if the current servo configuration in the firmware is neither 1520 nor 760.
  - If custom is active, all other fields are disabled
- Center (arm levelling) trim: -20.0% .. 20.0% (trim for correcting the center position)
- Max/Up trim: -20.0%..20.0% (trim for correcting the up position)
- Min/Down trim: -20.0%..20.0% (trim for correcting the down position)
- Max/Up travel: 50.0%..150.0%. (maximum is up direction when mounted on the heli)
- Min/Down travel: 50.0%..150.0% (minimum is down direction when mounted on the heli)
- Update rate: 50Hz..480Hz (1520us) or 50Hz..960Hz (760us)
- Servo reverse toggle
- Geometry Correction toggle (for rotational servos)

egonl, 2023-01-25:
I don't like the idea of imposing limitations to what the user can specify in the Configurator. I think it should only give warnings when values are unusual.

rotorflight, 2023-04-13:
Job done.

closed 2023-04-13 by rotorflight