rotorflight/rotorflight-configurator #25: Refactor Adjustments tab

rotorflight, 2022-12-28:
The Adjustment function has been totally refactored, and is working very differently. The Adjustments tab needs a similar refactoring.

- Adjustment type dropbox
  - Stepped / Continuous
  - Maybe combine with OFF switch... Off / Stepped / Continuous
- Enable channel (like before)
  - Add "Always enabled" in the list
- Control channel 
  - New dropdown menu and an slider below the enable channel
  - Two ranges needed with stepped adjustment
  -  One range needed with continuous adjustment
- Step size field as before
- Min & max value as before
- Adjustment function as before

Show the resulting adjustment value!

egonl, 2023-01-28:
@rotorflight Couple of questions:
- What should the OFF switch do?
- How would you implement 'Always enabled', could you give a CLI example?
- Would it be possible to use non-AUX channels as well? So you can adjust values with for example roll?

rotorflight, 2023-11-29:
Job done

closed 2023-11-29 by rotorflight