rotorflight/rotorflight-configurator #60: Receiver tab issue [snapshot 20230702]

rtlopez, 2023-07-22:
When enter to receiver tab configurator stops working. After any change, button "save" desn't appear, also I cannot switch to any other tab and reconnect is required.

I've tested with STM32F405 firmware (snapshot 0628) flashed to OMNIBUSF4SD board with this config https://github.com/rotorflight/rotorflight-targets/blob/master/configs/AIRB-OMNIBUSF4SD.config

After running devel branch of configurator, I'm getting this errors on console
```
receiver.js:423 Uncaught TypeError: Cannot read properties of undefined (reading 'val')
    at setRcMapGUI (receiver.js:423:35)
    at HTMLDivElement.process_html (receiver.js:542:9)
    at HTMLDivElement.<anonymous> (jquery.min.js:2:85239)
    at Function.each (jquery.min.js:2:3003)
    at S.fn.init.each (jquery.min.js:2:1481)
    at Object.<anonymous> (jquery.min.js:2:85221)
    at c (jquery.min.js:2:28327)
    at Object.fireWith [as resolveWith] (jquery.min.js:2:29072)
    at l (jquery.min.js:2:79901)
    at XMLHttpRequest.<anonymous> (jquery.min.js:2:82355)
```
and if I change channel order preset
```
receiver.js:423 Uncaught TypeError: Cannot read properties of undefined (reading 'val')
    at setRcMapGUI (receiver.js:423:35)
    at HTMLInputElement.<anonymous> (receiver.js:457:13)
    at HTMLInputElement.dispatch (jquery.min.js:2:43064)
    at HTMLInputElement.v.handle (jquery.min.js:2:41048)
    at Object.trigger (jquery.min.js:2:71515)
    at HTMLInputElement.<anonymous> (jquery.min.js:2:72110)
    at Function.each (jquery.min.js:2:3003)
    at S.fn.init.each (jquery.min.js:2:1481)
    at S.fn.init.trigger (jquery.min.js:2:72086)
    at S.fn.init.S.fn.<computed> [as change] (jquery.min.js:2:88570)
    at HTMLSelectElement.<anonymous> (receiver.js:463:47)
    at HTMLSelectElement.dispatch (jquery.min.js:2:43064)
    at HTMLSelectElement.v.handle (jquery.min.js:2:41048)
```

default firmware map shown in CLI is
```
map AETRC123
```

I'm not sure if it matters, I'm using Linux build of configurator.

rotorflight, 2023-07-23:
The title says "snapshot 20230702", but later says "After running devel branch".

Can you report which exact version of devel branch is doing this?
There have been lots of changes around this area, so if you did not test the latest, it may have been already fixed.

rotorflight, 2023-07-23:
Issue identified and fixed. Will appear in the next snapshot.

rtlopez, 2023-07-23:
That's great ;) I tested with `69a805e54ae342241701f9c8afa8b020f26f7657`. I can confim, if I set `serialrx_provider = SBUS` , thet receiver tab works correctly. Switching back to DSMX/1024 cause error again.

closed 2023-10-11 by rtlopez