rotorflight/rotorflight-configurator #108: HoTT ASCII telemetry (PID and other settings)

Ralfde, 2024-06-16:
feature request:
HoTT telemetry (ASCII/text mode):
betaflight and inav does support the settings f.e. PIDs, which are also supported by OSD to be adjusted to be adjusted in the HoTT telemetry ASCII/text mode.
Why was this function canceled at rotorflight?
Could this function be added, please?

AndersHoglund, 2024-10-31:
This is a firmware function, not RFC. 
HoTT Text mode CMS display port is included in firmware, it has not been "cancelled" or removed. If there is a problem with it, report as a firmware issue.

Ralfde, 2024-11-01:
Thanks a lot for the answer.After activating the CMS function it workes.

AndersHoglund, 2024-11-01:
OK, good. Then if you are happy, please close this issue.
/A

Ralfde, 2024-11-03:
Thanks a lot for your help again.

closed 2024-11-03 by Ralfde