rotorflight/rotorflight-configurator #196: bug: tail RPM is not displayed correctly when using RPM sensor

egonl, 2024-11-10:
Configuration:
- Main motor uses bidirectional DSHOT
- Tail motor uses an RPM sensor on resource FREQ 2

In this configuration the Configurator shows the main motor RPM instead of the tail motor RPM. 

Example: here the RPM sensor is connected to motor 1 - while it's configured for motor 2 (reason: I wanted to compare DSHOT RPM telemetry with an RPM sensor):

[image]

heliflieger, 2026-11-07:
Looking at this from the Configurator side: the tail RPM is read from the motor telemetry entry of the tail motor (MSP_MOTOR_TELEMETRY, motor 2 when the tail mode is motorised) and scaled by the tail gear ratio. The index itself matches the setup described here.

The Ethos suite takes the tail speed from the flight controller's own telemetry sensor "Tailspeed" instead (lib/live_sensors.lua), so the two clients may not see the same value for an RPM sensor on FREQ 2.

To find out which side is wrong, could you share:
- the MSP_MOTOR_TELEMETRY reply (the rpm value of each motor entry) while the RPM sensor shows 767 and the Configurator shows 775,
- the Tailspeed telemetry value, if your receiver logs it,
- the firmware version and the motor and sensor setup (which motor the FREQ sensor is assigned to).

Not changed in the Configurator yet, pending that data.

egonl, 2026-11-08:
This is not a question about which side is wrong, but merely a display bug. The Configurator shows the correct value under *Motor #2 - Tail Motor* (767). Only in the summary view on top it displays the wrong value for *Tail Motor* (775).