rotorflight/rotorflight-configurator #285: Manual flash without target is allowed

plesogor, 2025-03-22:
In testing some master/PR builds i found out that you can flash a FBL with a hex file without a corresponding target for remapping. This puts the FBL into a NO_GYRO state arming flag, and the board is labeled as "BARE" in RFC.  

To produce download a hex, use RFC update firmware tab, select manual upload, and do not touch the target/hardware dropdowns. Just flash. 

My idea was to gray out / block the "Flash Firmware" button until Target is filled out with values, or auto-detection does it for you.  This will prevent this in the future when doing manual flashes.

rtlopez, 2025-03-22:
Developers need this.

plesogor, 2025-03-22:
makes sense dev boards, and targets not available for devs.

closed 2025-03-22 by plesogor

rtlopez, 2025-03-22:
Nevetheless I think it is worth to consider to provide spacial "empty" target for this purpose and do it as you suggest. This would allow to avoid confusion of other users if they do it accidentally.