rotorflight/rotorflight-configurator #344: RF Configurator - Font and UI  MAX Zoom  in Windows

toratoza, 2025-07-11:
Hi,

New to RF and RM Nexus , I was Configuring the motor - during enable motor override

Sliding the percentage bar to turn the motor and notice the configurator display started to zoom out ( uncontrollable ) causing the font and UI to be too small . Now I cannot restore the Configurator display to its original aspect or ratio. 

Did uninstalled RF Configurator 2,2  but no luck. 
 
Appreciate help or advice to get back normal display of RF Configurator. 

Thank you.  

[image]

breavyn, 2025-07-11:
Zoom is adjusted by scrolling in/out while holding down Ctrl.

toratoza, 2025-07-11:
> Zoom is adjusted by scrolling in/out while holding down Ctrl.

That did it ! Scrolling the mouse wheel in/out while holding down the ctrl button. 

Thks .